3DiVi/nuitrack-sdk #1: Missing UNITY_ANDROID in NuitrackManager.cs

jesusatgar, 2019-11-11:
There is a bug in NuitrackManager.cs in the Unity plugin.
In line num. 195, instead of if UNITY_IOS, it should be:
`#if UNITY_IOS || UNITY_ANDROID
		if (wifiConnect == WifiConnect.VicoVR)
		{
			nuitrack.Nuitrack.Init("", nuitrack.Nuitrack.NuitrackMode.DEBUG);`

This led to some problems while using the plugin for my Android application (Nuitrack wasn't correctly identifying the platform, and thus it tried to load the Windows .dll files, which obviously weren't installed).

I suggest updating the Unity examples/tutorials, since all of them require some tweaking or modifying code before being usable. Still I haven't managed to make the SegmentExample to work.

olga-komyakova, 2019-11-19:
Hi jesusatgar,

1. Are you trying to run the project from the Unity editor with Android set as target platform? 
2. If you run the project on an Android device, you have to install the VicoVR Manager (there's a setting for Wi-Fi connection). You don't have to make any changes in NuitrackManager. Nuitrack with iOS is an exceptional case, that's why there's a condition in NuitrackManager.

jesusatgar, 2019-11-19:
I am running the project from my Android device. Skeleton tracking works
just fine, everything has been configured in the VicoVR as explained in the
tutorials.

olga-komyakova, 2019-11-19:
Thanks for your feedback. Do you have any other questions about this case? Looking forward to your update.

Ivan-3DiVi, 2020-10-12:
Fixed in https://github.com/3DiVi/nuitrack-sdk/releases/tag/v0.34.0

closed 2020-10-12 by Ivan-3DiVi