3DiVi/nuitrack-sdk #2: Please provide packages (or other means for installing) for RPM-based distributions such as Fedora

esbennn, 2020-04-21:
It's great that you provide a .deb package for Ubuntu.
However, it would be even better if you also provided an RPM for linux user on other distributions, such as Fedora, OpenSuse, Arch etc. A generic tarball could be used for any distribution.

VadimAspirin, 2021-11-26:
Hi @esbennn, it’s in our roadmap, but so far without an exact implementation date, primarily due to low demand.

andrey-tsb, 2022-03-24:
(Just as an information for interested users / customers)
This enhancement (like many others) is at low-priority in backlog due to limited demand.
But if you have some interest in enhancing Nuitrack in some specific/custom way and have a volume demand in terms of licenses - just let us know, we are always open for discussion to meet our customers needs.

closed 2023-01-26 by andrey-tsb