3DiVi/nuitrack-sdk #3: Garbage collection leak on Nuitrack.NET 0.30

vpenades, 2020-06-24:
Last year I reported in nuitrack forums an issue about Nuitrack.net creating new large arrays for every new frame, which made the GC to trigger a garbage collection every few frames.

I provided a solution, and Nuitrack was kind enough to send us a private library which had this problem solved.

We've upgraded to Nuitrack 0.30, which requires a new Nuitrack.NET dll, and after trying it, we've discovered it still has the same garbage collection problem from a year ago.

I can't understand why, a bug that was reported, aknowledged and FIXED is still not being released to the public.

Please, fix it ASAP and release a new nuitrack.NET dll immediately.

olga-komyakova, 2020-06-26:
Hi @vpenades,

Sorry for the issues you've encountered.

This problem will be fixed in the new release, which is planned next week.

vpenades, 2020-06-26:
@olga-komyakova Hi again, given that you're going to update the nuitrack.net.dll for the next version, I would like you to take these into consideration:

1. The Byte[] array that nuitrack.net.dll exposes is designed to make things easy for the end developer, in exchange for a performance loss. This is addressed by the fix I proposed a year ago. BUT, if the API exposes the internal pointer as an IntPtr, we can take advantage of that and avoid a memory block copy entirely, which would be a great performance improvement.

2. We're in the process of moving our projects to NetCore 3 , I am not sure which framework nuitrack.net.dll uses, but if it is one of the old NetFramework 3.5, or 4.5 etc, you should consider exploring how to make it work in NetCore..... most probably, you need to compile the wrapper targeting NetStandard2.0 or something like that.  Right now there's probably a lot of clients still using NetFramework.... but everybody is moving towards NetStandard and NetCore.... so you should take that into account and update the framework accordingly.

olga-komyakova, 2020-07-02:
We've released the new version, you can check it out.

vpenades, 2020-07-02:
Hi @olga-komyakova I've tested the latest version and I can see the IntPtr data is now exposed, which is much more convenient since it avoids doing a frame copy.

For those that still require a plain old c# array, it can be done in this way (requires c# unsafe mode):

```c#
var data = new Span<Byte>(frame.Data.ToPointer(), frame.DataSize);
var array = data.ToArray();
```

Thanks for fixing this issue!

closed 2020-07-02 by vpenades