3DiVi/nuitrack-sdk #4: L515 not working in Unity 0.34.1 (at least for me)

sercasyr, 2020-11-08:
I have an Intel Lidar L515 sensor and when I try the demos of NuiTrack for Unity (for instance the BasicSkeleton demo) I get a ModuleNotInitializationException. It does not recognise the device.

I am using the latest version of the Unity package (0.34.1) on Unity 2019.4.13f1.  I use a USB3 port.
NuiTrack demos work fine with an Intel RealSense 435i.
The Intel L515 works with the RealSense Viewer of Intel RealSense SDK, but it does not work using NuiTrack.
As the 0.34.1 is a very recent version, I wonder if there is a bug, because the 0.34.1 is expected to support the L515.

olga-komyakova, 2020-11-09:
Hi @sercasyr,

Sorry for the issues you've encountered. Please advise did you update the Nuitrack runtime to v0.34.1 (besides updating the Unity package)?

sercasyr, 2020-11-09:
Sorry, I think I didn't do that. 
That must be the cause.

closed 2020-11-09 by sercasyr