3DiVi/nuitrack-sdk #6: Qt_5.11 not found during activation

bearpaw, 2020-12-25:
Hi Nuitrack team,

I tried to follow the instructions to run nuitrack. However, I got this error:
```
/usr/bin/QtNuitrackLicense: /usr/lib/x86_64-linux-gnu/libQt5Core.so.5: version `Qt_5.11' not found (required by /usr/bin/QtNuitrackLicense)
```

My system is Ubuntu 18.04: 
`5.4.0-53-generic #59~18.04.1-Ubuntu SMP Wed Oct 21 12:14:56 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
`
Any help would be appreciated! Thanks!

VadimAspirin, 2021-11-26:
Please advise, do you have this error with the latest Nuitrack version (v0.35.11)?

VadimAspirin, 2022-03-31:
Hi @bearpaw,

How are you? Has your issue been solved? 
Would be great if you could provide some reply/feedback, we will be ready to help.
Thanks.

closed 2022-04-14 by VadimAspirin

crystal-butler, 2022-05-11:
I have the same issue under Ubuntu 18.04. I downloaded the nuitrack installer just now, so presumably have the latest version of the application.