3DiVi/nuitrack-sdk #7: Terminate Exception I got an issue with Unity

o986o, 2021-01-29:
[image]

Aupuma, 2021-02-01:
Same issue here, using Unity 2020.1.0f1

reuksv, 2021-04-07:
https://github.com/3DiVi/nuitrack-sdk/blob/master/doc/Install.md#nuitrack-plugins  
* close the Unity editor;
* replace the TBB library in your Unity editor with the library from Nuitrack:
* Windows: copy the tbb.dll library from %NUITRACK_HOME%/bin to your Unity editor folder (for example, C:\Program Files\Unity\Hub\Editor\2019.4.8f1\Editor);
Linux: copy the libtbb.so.2 library from /usr/local/lib/nuitrack/ to your Unity editor folders (for example, /home/user/2019.3.3f1/Editor and /home/user/2019.3.3f1/Editor/Data/Tools);
run the Unity editor again.

Trutifruti, 2021-04-21:
@reuksv i did what you advised but the error persists, would you have any idea  why ?

reuksv, 2021-05-07:
@Trutifruti Very strange. Which version of nuitrack, unity, and also which sensor and OS are you using?