3DiVi/nuitrack-sdk #8: Assertion failed UnityEngine.GUIUtility:ProcessEvent (int,intptr)

match08, 2021-03-14:
Assertion failed
UnityEngine.GUIUtility:ProcessEvent (int,intptr)

EugeneVasiliev, 2021-04-07:
This is a Unity system issue.
Try making Reimport all or using a different version of Unity.

<details>  

[image]  
</details>

This may be useful: [(forum.unity.com) UnityEngine.GUIUtility:ProcessEvent(Int32, IntPtr)](https://forum.unity.com/threads/unityengine-guiutility-processevent-int32-intptr.514797/#post-6935341)

Alexander-Shamin, 2021-04-07:
Hello! I have the same problems.

What i did:
- install Unity3d 2019.4.4f1, Unity 2020.3.2f1
- install Nuitrack v.035.2, v.035.0, v.035.3.
- create different project for all combination of Unity andd Nuitrack.

Using system variable switched nuitrack sdk (reboot system between switched).
In all project have "Assertion failed"

Also i have build, with Nuitrack sdk. After update version v.35.2 my application start failed. And unity don't have any messages or errors.

EugeneVasiliev, 2021-04-07:
Dear, Alexander Yu Shamin!

We are working on studying this problem and finding a solution.

> works nearly 1-2 hour and after that failed  

How exactly did Unity fail? Crash, Nuitrack shutdown without stopping the editor, or something else?

> Also I use you api for gender detection - and sometime you publish non complete Json file.

If any parameters cannot be defined with the specified accuracy, they will not be [included in the Json](https://github.com/3DiVi/nuitrack-sdk/blob/master/doc/Instance-based_API.md).

Learn more from the example: [Face Tracking with Nuitrack](https://github.com/3DiVi/nuitrack-sdk/blob/master/doc/Unity_Face_Tracking.md)

Alexander-Shamin, 2021-04-07:
> works nearly 1-2 hour and after that failed

In build version -> Unity application stops - without any messages (log, warning, error). In players logs no information. No Unity exception or assert. But unity events OnExit, OnDestroy were not call.

Editor version -> work fine, the application has worked for more than 24 hours

> Also I use you api for gender detection - and sometime you publish non complete Json file.
What i mean - you publish json string and this string parse it
```
        string json = nuitrack.Nuitrack.GetInstancesJson();
	// Replace quotation marks with square brackets to prevent a conversion error 
	// in case an array is empty (no info about faces received)
	faceInfo = JsonUtility.FromJson<FaceInfo>(json.Replace("\"\"", "[]"));
```

But, sometime i have exception on parsing string `ArgumentException: JSON parse error: Missing a comma or '}' after an object member.`
I check the json string -> it breaks off. 
The exception's rate is not deterministic.

Thank you!

Alexander-Shamin, 2021-04-07:
Publish full assert message:
```
Assertion failed
0x00007FF726C89E8C (Unity) StackWalker::GetCurrentCallstack
0x00007FF726C8DE41 (Unity) StackWalker::ShowCallstack
0x00007FF725386225 (Unity) GetStacktrace
0x00007FF72790947E (Unity) DebugStringToFile
0x00007FF726CE91B5 (Unity) DebugLogHandler_CUSTOM_Internal_Log
0x0000019FC00E70EB (Mono JIT Code) (wrapper managed-to-native) UnityEngine.DebugLogHandler:Internal_Log (UnityEngine.LogType,UnityEngine.LogOption,string,UnityEngine.Object)
0x0000019FC00E6CBB (Mono JIT Code) UnityEngine.DebugLogHandler:LogFormat (UnityEngine.LogType,UnityEngine.Object,string,object[])
0x0000019FC00E613E (Mono JIT Code) UnityEngine.Logger:Log (UnityEngine.LogType,object)
0x0000019FC04196BB (Mono JIT Code) UnityEngine.Debug:Assert (bool)
0x0000019FCFCECA2B (Mono JIT Code) UnityEngine.UIElements.UIR.Implementation.RenderEvents:DepthFirstOnClippingChanged (UnityEngine.UIElements.UIR.RenderChain,UnityEngine.UIElements.VisualElement,UnityEngine.UIElements.VisualElement,uint,bool,bool,bool,bool,bool,UnityEngine.UIElements.UIR.UIRenderDevice,UnityEngine.UIElements.UIR.ChainBuilderStats&)
0x0000019FCFCED6D3 (Mono JIT Code) UnityEngine.UIElements.UIR.Implementation.RenderEvents:DepthFirstOnClippingChanged (UnityEngine.UIElements.UIR.RenderChain,UnityEngine.UIElements.VisualElement,UnityEngine.UIElements.VisualElement,uint,bool,bool,bool,bool,bool,UnityEngine.UIElements.UIR.UIRenderDevice,UnityEngine.UIElements.UIR.ChainBuilderStats&)
0x0000019FCFCED6D3 (Mono JIT Code) UnityEngine.UIElements.UIR.Implementation.RenderEvents:DepthFirstOnClippingChanged (UnityEngine.UIElements.UIR.RenderChain,UnityEngine.UIElements.VisualElement,UnityEngine.UIElements.VisualElement,uint,bool,bool,bool,bool,bool,UnityEngine.UIElements.UIR.UIRenderDevice,UnityEngine.UIElements.UIR.ChainBuilderStats&)
0x0000019FCFCED6D3 (Mono JIT Code) UnityEngine.UIElements.UIR.Implementation.RenderEvents:DepthFirstOnClippingChanged (UnityEngine.UIElements.UIR.RenderChain,UnityEngine.UIElements.VisualElement,UnityEngine.UIElements.VisualElement,uint,bool,bool,bool,bool,bool,UnityEngine.UIElements.UIR.UIRenderDevice,UnityEngine.UIElements.UIR.ChainBuilderStats&)
0x0000019FCFCED6D3 (Mono JIT Code) UnityEngine.UIElements.UIR.Implementation.RenderEvents:DepthFirstOnClippingChanged (UnityEngine.UIElements.UIR.RenderChain,UnityEngine.UIElements.VisualElement,UnityEngine.UIElements.VisualElement,uint,bool,bool,bool,bool,bool,UnityEngine.UIElements.UIR.UIRenderDevice,UnityEngine.UIElements.UIR.ChainBuilderStats&)
0x0000019FCFCED6D3 (Mono JIT Code) UnityEngine.UIElements.UIR.Implementation.RenderEvents:DepthFirstOnClippingChanged (UnityEngine.UIElements.UIR.RenderChain,UnityEngine.UIElements.VisualElement,UnityEngine.UIElements.VisualElement,uint,bool,bool,bool,bool,bool,UnityEngine.UIElements.UIR.UIRenderDevice,UnityEngine.UIElements.UIR.ChainBuilderStats&)
0x0000019FCFCEBAA3 (Mono JIT Code) UnityEngine.UIElements.UIR.Implementation.RenderEvents:ProcessOnClippingChanged (UnityEngine.UIElements.UIR.RenderChain,UnityEngine.UIElements.VisualElement,uint,UnityEngine.UIElements.UIR.UIRenderDevice,UnityEngine.UIElements.UIR.ChainBuilderStats&)
0x0000019FCFCE6683 (Mono JIT Code) UnityEngine.UIElements.UIR.RenderChain:Render (UnityEngine.Rect,UnityEngine.Matrix4x4,UnityEngine.UIElements.PanelClearFlags)
0x0000019FCFCE56AB (Mono JIT Code) UnityEngine.UIElements.UIRRepaintUpdater:DrawChain (UnityEngine.Rect,UnityEngine.Matrix4x4)
0x0000019FCFCE463E (Mono JIT Code) UnityEngine.UIElements.UIRRepaintUpdater:Update ()
0x0000019FC062FA2A (Mono JIT Code) UnityEngine.UIElements.VisualTreeUpdater:UpdateVisualTreePhase (UnityEngine.UIElements.VisualTreeUpdatePhase)
0x0000019FCFCE2513 (Mono JIT Code) UnityEngine.UIElements.Panel:UpdateForRepaint ()
0x0000019FCFCE1BCB (Mono JIT Code) UnityEngine.UIElements.Panel:Repaint (UnityEngine.Event)
0x0000019FC062E7D1 (Mono JIT Code) UnityEngine.UIElements.UIElementsUtility:DoDispatch (UnityEngine.UIElements.BaseVisualElementPanel)
0x0000019FC062E14B (Mono JIT Code) UnityEngine.UIElements.UIElementsUtility:ProcessEvent (int,intptr)
0x0000019FC062D9B3 (Mono JIT Code) UnityEngine.GUIUtility:ProcessEvent (int,intptr)
0x0000019FC062DC33 (Mono JIT Code) (wrapper runtime-invoke) <Module>:runtime_invoke_bool_int_intptr (object,intptr,intptr,intptr)
0x00007FFCDF61D690 (mono-2.0-bdwgc) [mini-runtime.c:2809] mono_jit_runtime_invoke
0x00007FFCDF5A2912 (mono-2.0-bdwgc) [object.c:2921] do_runtime_invoke
0x00007FFCDF5AB96F (mono-2.0-bdwgc) [object.c:2968] mono_runtime_invoke
0x00007FF726BFDDBE (Unity) scripting_method_invoke
0x00007FF726BF7AFD (Unity) ScriptingInvocation::Invoke
0x00007FF726BF1B0A (Unity) ScriptingInvocation::Invoke<bool>
0x00007FF723F578E5 (Unity) Scripting::UnityEngine::GUIUtilityProxy::ProcessEvent
0x00007FF7248B15F4 (Unity) GUIView::ProcessRetainedMode
0x00007FF7253ACCD9 (Unity) GUIView::OnInputEvent
0x00007FF7248B14EC (Unity) GUIView::ProcessInputEvent
0x00007FF7253A317D (Unity) GUIView::DoPaint
0x00007FF7253AF74D (Unity) GUIView::RepaintAll
0x00007FF724A45723 (Unity) PlayerLoopController::UpdateScene
0x00007FF724A2E42A (Unity) PlayerLoopController::EnterPlayMode
0x00007FF724A40A05 (Unity) PlayerLoopController::SetIsPlaying
0x00007FF724A4283B (Unity) Application::TickTimer
0x00007FF7253ABA1E (Unity) MainMessageLoop
0x00007FF7253B5C58 (Unity) WinMain
0x00007FF7283A7792 (Unity) __scrt_common_main_seh
0x00007FFD56BE7C24 (KERNEL32) BaseThreadInitThunk
0x00007FFD5774D4D1 (ntdll) RtlUserThreadStart
(Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 35)
```

EugeneVasiliev, 2021-04-07:
I have a couple of clarifying questions:

1.  Did I understand correctly that this Assert occurs ONLY in projects with Nuitrack?

1.  Does Assert occur when the project is opened or at runtime?

andrey-tsb, 2021-04-07:
> Если есть желание - можем с вами связаться, я могу дать детальный апдейт об последней версии. Я более года использую ваше sdk для своего продукта, и пробовал перейти уже на AI версию - столкнулся с множеством вопросов.
> Можем связаться в телеграмм если интересно - @alexander-yu-shamin

Александр, в дополнение к текущему треду в ближайшие дни обязательно постучимся к вам в telegram. Обратная связь по AI версии была бы очень нам полезна.

Alexander-Shamin, 2021-04-07:
> I have a couple of clarifying questions:
> 
> 1. Did I understand correctly that this "Assertion failed" occurs ONLY in projects with Nuitrack?
> 2. Does Assert occur when the project is opened or at runtime?

1. Both type of project. And in clear project with only nuitrack and in exist project. 
2. Only in Editor.

I don't try run you preview project more than 15 minutes, because can't say have they some problems, that my project.

EugeneVasiliev, 2021-05-11:
> Both type of project. And in clear project with only nuitrack and in exist project.

Do projects without Nuitrack have this problem?  This looks like a problem in your system or Unity.

> But, sometime i have exception on parsing string ArgumentException: JSON parse error: Missing a comma or '}' after an object member.  

Yes, we also found this problem. As a temporary solution before the fixes in Nuitrack, you can wrap the code:
`faceInfo = JsonUtility.FromJson<FaceInfo>(json.Replace("\"\"", "[]"));` in try - catch.

Could you provide json with an error for comparison? To catch it, add this code, and then provide the console output.

```
try
{
    faceInfo = JsonUtility.FromJson<FaceInfo>(json.Replace("\"\"", "[]"));
}
catch
{
     Debug.Log(json);
}
```

Waiting for your reply.

EugeneVasiliev, 2021-06-01:
Try enabling asynchronous initialization of Nuitrack, this should fix the problem with "Assertion failed".

<details>

[image]
</details>