3DiVi/nuitrack-sdk #10: nuitrack sdk demo can't open intel realsense D435

AprilCat, 2021-03-24:
Hello,
    I ran nuitrack sdk demo, but demo ran failed. there are some information in console:
=====================================================================
Usage: nuitrack_console_sample [path/to/nuitrack.config]
Can't find video camera!
Can't find Depth camera!
ERROR: Can't create DepthSensor module
AstraProPerseeDepthProvider: Can't open device (	DeviceOpen using default: no devices found
)
KinectDepthProvider: Devices not found
OpenNI2DepthProvider: Can't open device (	DeviceOpen using default: no devices found
)
Realsense2DepthProvider: Can't initialize Realsense2: rs2_config_resolve(config:0x563d423b4500, pipe:0x563d423aa800):
    Couldn't resolve requests

terminate called after throwing an instance of 'tdv::nuitrack::ModuleNotInitializedException'
  what():  

Process finished with exit code 134 (interrupted by signal 6: SIGABRT)
=====================================================================
Can anyone tell me how to solve it? Thanks.
BTW, I use command nuitrack to test camera, it shows error.
[image]

reuksv, 2021-05-11:
Hello. Is there an error if you turn off all these checkboxes?
[image]

BlindElephants, 2022-04-27:
I'm having this same problem. Fails if all three are checks in the Nuitrack executable, and also fails when trying to initialize skeleton tracking from the Python API. I am working with a Pro license. 

From the Python API, this is the error:
```
ERROR: Can't create DepthSensor module
Realsense2DepthProvider: Can't initialize Realsense2: rs2_config_resolve(config:00000232A094CD50, pipe:00000232A0AB9C40):
    Couldn't resolve requests
```

Please advise

koktavy, 2024-12-23:
@reuksv I have the same error:
```
Can't create DepthSensor module
Realsense2DepthProvider: Can't initialize Realsense2: rs2_config_resolve(config:0000000013516670, pipe:0000000013517630):
    Couldn't resolve requests
```
It happens in classical or AI mode, with AI features on or off, error reporting on or off.

@AprilCat @BlindElephants did you ever find a solution?

koktavy, 2024-12-23:
I have a positive update, turns out my USB drivers were out of date. In the RealSense Viewer all ports were reporting USB 2.1 connection when I definitely have 3.0 ports on this PC.

To fix the issue I started in Device Manager. Across all USB entries I tried to right click -> update driver -> search automatically. "Best drivers already installed" for everything. BUT I clicked "Search for updated drivers on Windows Update" and did find optional updates that I hadn't seen before.

After installing those, the RealSense Viewer now reports a 3.2 connection and the Nuitrack demo opens successfully. Note—I also tried various other cables and **only** the official Intel cable works, as mentioned elsewhere online.

reuksv, 2024-12-23:
Hello @koktavy 
I also want to draw your attention to the fact that problems may arise depending on which side the type-c cable was plugged in. Depending on the plug rotation, the displayed version of USB may be unstable and change from launch to launch. If you turn the plug (type-c) 180 degrees, you can get a consistently good connection.