3DiVi/nuitrack-sdk #12: Unity3D and Android on Persee/TVico crash

Vangos, 2021-04-16:
## Scenario

I am using the Orbbec Persee/TVico camera to stream color and skeleton frames on an Android app.

I've purchased a license and then followed the steps in your [Wireless case](https://github.com/3DiVi/nuitrack-sdk/blob/master/doc/TVico_User_Guide.md#wireless-case) documentation to setup the Persee device. Persee is properly running as a WiFi hotspot.

Also, I downloaded the VicoVR.apk and assigned the proper IP address, exactly as described in your [Android section](https://github.com/3DiVi/nuitrack-sdk/blob/master/doc/TVico_User_Guide.md#android). When I open the "Test Sensor" screen on the VicoVR app, everything works fine.

## The issue

I downloaded the [latest Unity package](https://github.com/3DiVi/nuitrack-sdk/blob/master/Unity3D/NuitrackSDK.unitypackage) from your repository. When I build a Unity app for Android, the app is crashing on Android. Here's how to reproduce:

- Create a new Unity project and import the Nuitrack Unity package. Also, the Unity installation folder includes the proper tbb.dll.
- Open `NuitrackManager.cs` and assign the correct IP address of the Persee camera.
- Open the `RGBandSkeletons` Unity scene (or any other built-in scene).
- In the `NuitrackScripts`, set the **Wifi Connect** option to **TVico** (also tried with **VicoVR**).
- Run in the Editor - works OK.
- Build for Android and run.
- **The application starts, displays the Unity logo, and immediately crashes.**

Here's the output of the Android Logcat:

```
nuitrack.ModuleNotInitializedException
   at nuitrack.Nuitrack.Update ()
   at NuitrackManager.Update ()
``` 

## Notes

- Tried on 2 different Android phones.
- The Persee camera, Windows computer, and Android device are all connected to the same network.
- I've used the Nuitrack/Auto switch dll option.
- I've tried to set the Persee as a WiFi hotspot and as a Portable hotspot, leaving the default IP address. Same behavior.
- Additionally, I've tried various Unity packages from the repo History. Same behavior.

reuksv, 2021-04-16:
Hello. 
It's steps only for TVico + PC
> Open NuitrackManager.cs and assign the correct IP address of the Persee camera. 
> In the NuitrackScripts, set the Wifi Connect option to TVico (also tried with VicoVR).

If you need TVico + android try with "none"
[image]

Vangos, 2021-04-16:
Thank you for your prompt respons. It is still crashing on the Android app.

Here's what we did:

- In `NuitrackManager.cs`, set the IP address of the Persee camera.
- In the Editor, the **WiFi Connect** option is set to **None**.
- Built and run on the Android phone.
- The app crashed.

_We can install the Android app directly into the Persee camera and works with no issues there. However, we are interested in the Wireless case -- not the Standalone case._

reuksv, 2021-04-20:
Hm, I tested on last VicoVR.apk and last TVico.apk, and it works.
If testing in vicovr app works fine. Please try our Unity Demo https://play.google.com/store/apps/details?id=com.VicoVR.UnityDemo&hl=ru
If the demo works, check Player Settings. Make sure the settings look the same. And then try to build AllModulesScene
[image]

P.S. For your case, `NuitrackManager.cs`  does not need to be edited.

Vangos, 2021-04-21:
We copied your settings and it worked. Thank you for your support. Feel free to close this ticket.

> P.S. For your case, NuitrackManager.cs does not need to be edited.

Just out of curiosity, if we don't edit the IP field, how is it going to work on a local network?

reuksv, 2021-04-21:
It only needs to work TVico + PC (or iOS). This assumes the local IP of TVico is always 192.168.43.1

closed 2021-04-21 by reuksv