3DiVi/nuitrack-sdk #13: Can i apply your code to my desktop

Trutifruti, 2021-04-16:
Hi, first of all great work ! I wanted to know if it's possible to use your code to devellope a touchless UI on my computer ? The goal being to be able to rigth/left click and move the mouse around on my pc in general.

olga-komyakova, 2021-04-16:
Hi @Trutifruti! Thanks for your feedback! Yes, you can use Nuitrack SDK for your use case. Take a look at the Hand Tracker and Gesture Recognizer modules in our [API documentation](https://download.3divi.com/Nuitrack/doc/modules.html). You can also check out our [Unity tutorial on hand tracking](https://github.com/3DiVi/nuitrack-sdk/blob/master/doc/Unity_Gallery.md). Please note that you'll need a 3D sensor (see the list of [compatible sensors and platforms](https://nuitrack.com/#sensors)) and installed [Nuitrack runtime](https://github.com/3DiVi/nuitrack-sdk/blob/master/doc/Install.md). For testing purposes, you can use the [trial license](https://nuitrack.com/#pricing).

eddyaboumansour, 2021-04-18:
@olga-komyakova Thank you for your help. If I want to apply the first part of this code (the one where we set the Pointer Control without the gallery)  to control Google Earth for example on Chrome browser or any web page or even desktop apps (word, excel), could you give me a starting point on how to do this? (Sorry I'm new to unity).

Trutifruti, 2021-04-19:
Thank you so much @olga-komyakova for your great tips I’m going to get right to it then!

closed 2021-04-23 by Trutifruti