3DiVi/nuitrack-sdk #14: Error while activating the SDK

lassolis, 2021-04-17:
Hello and thank you for this great SDK.
But i have one problem while use the activation tool, when i i use activation_tool/Nuitrack.exe it just doesn't lunch, i checked the event viewer it showed that there is dll problems. First there was the libstdc++-6.dll, so i downloaded it and copy it to the folder, and again when i try to lunch the activation tool it doesn't do anything, and then i checked the event viewer and it showed that there was an error with ntdll.dll from /windows/system32.
My platform is win 10 64bit.
I would like to get help on this issue, because i am unable to use the SDK in any unity project.
And thank you.

EugeneVasiliev, 2021-05-07:
Please check if the following software components are installed.

https://github.com/3DiVi/nuitrack-sdk/blob/master/doc/Install.md#software-requirements

reuksv, 2021-10-14:
Case closed due to no further comments received.