3DiVi/nuitrack-sdk #15: NtWaitForMultipleObjects exception

Alexander-Shamin, 2021-04-20:
Hello! 
I use you library in Unity3d with webcamera.
Sometime i have fail of application with error:
```
0x00007FFCC87FD244 (ntdll) NtWaitForMultipleObjects
0x00007FFCC5762AE7 (KERNELBASE) WaitForMultipleObjectsEx
ERROR: SymGetSymFromAddr64, GetLastError: 'Попытка обращения к неверному адресу.' (Address: 000000006185FD57)
0x000000006185FD57 (NuitrackModule) (function-name not available)
0x0000000061C9D974 (NuitrackModule) ZN5boost13serialization9singletonISt8multisetIPKNS0_18extended_type_infoENS0_6detail11key_compareESaIS5_EEE20get_mutable_instanceEv
ERROR: SymGetSymFromAddr64, GetLastError: 'Попытка обращения к неверному адресу.' (Address: 0000000061BD0140)
0x0000000061BD0140 (NuitrackModule) (function-name not available)
ERROR: SymGetSymFromAddr64, GetLastError: 'Попытка обращения к неверному адресу.' (Address: 000000006186167D)
0x000000006186167D (NuitrackModule) (function-name not available)
0x00007FFCC7FAB04A (msvcrt) beginthreadex
0x00007FFCC7FAB11C (msvcrt) endthreadex
0x00007FFCC6E07C24 (KERNEL32) BaseThreadInitThunk
0x00007FFCC87CD4D1 (ntdll) RtlUserThreadStart
```

For example log of application.
[error.log](https://github.com/3DiVi/nuitrack-sdk/files/6344148/error.log)

Alexander-Shamin, 2021-04-30:
I use realsense D415 for Nuitrack and Webcam for RGB image. (2 cameras at one time). Nuitrack fall in ZN5boost13serialization9singletonISt8multisetIPKNS0_18extended_type_infoENS0_6detail11key_compareESaIS5_EEE20get_mutable_instanceEv

reuksv, 2021-04-30:
Are there any memory leaks in the process?

Alexander-Shamin, 2021-04-30:
No.

reuksv, 2021-05-07:
Try to run a standard scene from a package. For example "allModulesScene". Will the application crash again?

Alexander-Shamin, 2021-05-07:
I can't run example on release machine, where number of people passing by per minute near 3-4. 
Also, i can run exmaple on 8 hours bag file - becouse your api can't do that.

So, i can't run example in simular conditions.

reuksv, 2021-05-13:
If you run it on a test machine, is there no problem?
Have you tried running a project without a webcam?
Can you send us a project (or a part of it that would cause the same error)?
I didn't quite understand about the bag file. You can use the recorded bag file (or .oni), just add the path to the file in Nuitrack\nuitrack\nuitrack\data\nuitrack.config
[image]

Alexander-Shamin, 2021-05-13:
With test machine i don't have any problems.
Yes, i tried - application fail.
No - i can't send project to you.
We can't use .bag file longer than 3 minutes. You have a very strangle version of license. I can't run with another version of serial number.

reuksv, 2021-05-20:
Then describe in more detail how your application interacts with nuitrack.
1. Does scene restart or nuitrack take place?
2. What type of initialization are you using (standard or asynchronous)?
3. Which API compatibility level is selected?
4. Graphics API?
5. Scripting backend?
6. Have you tried building on a different version of Unity (2018, 2020, 2021)?
7. Also try the newest nuitrack (v0.35.5)