3DiVi/nuitrack-sdk #18: Jittery hand tracking

Trutifruti, 2021-04-23:
Hi all! so I have followed the tutorial and when I get up to step 11 (two hand sprits moving on the background accordingly to my hand movements) and it works! However, there is some jitteriness to it all. from time to time the sprits have spasms and if I am too carless in my movements, it becomes quite hard to use. My co-worker advised that lowering the min velocity interactive point, which did help a bit. But is there an ideal set up for the camera or a configuration that I missed? The gifs in the tutorial seem to show a user much less constrained in his movements.
I am using the D435 for reference.  
Thanks and have a nice day!

EugeneVasiliev, 2021-05-11:
The quality of hand tracking depends on the resolution of your sensor. 
In the tutorial, we used the Orbbec Astra.
The D435 sensors, despite their higher depth resolution, have low tracking accuracy and noise, so you can't get results like in GIFs.

Try using other sensors [supported by Nuitrack](https://nuitrack.com/#sensors).

You can improve the result by experimenting with **nuitrack.config**.
To change the hand detection area, you can change the values: `HandTracker.HandPointerFrame.Width` \ `Height` in **nuitrack.config** (`<nuitrack-path>\nuitrack\nuitrack\data\nuitrack.config`).

Trutifruti, 2021-05-19:
@EugeneVasiliev i'm sorry I haven't tested your advise yet, we are working out other issues with my partners for the moment. However we will experiment with the nuitrack.config as soon as our database issues (not related to nuitrack) are solved.

Trutifruti, 2021-05-26:
@EugeneVasiliev  so we tried your advice but it didin't help to much, i think we just need to get à more adapted sensor, thanks for the tip !

closed 2021-10-13 by reuksv