3DiVi/nuitrack-sdk #19: Implementing HandsInput on unity-webview

eddyaboumansour, 2021-04-25:
Hello, first thank you for this tutorial it's very helpful but instead of a gallery I'm using a WebView that displays a webpage (Google Earth globe). But Mouse, Move, Drag processes are not working on the WebView, what seems to be the problem? Aren't the pointers acting as a mouse already? And aren't the OnClick events of a webpage already implemented ? I would love a small guide on how to achieve this solution. Thank you!

EugeneVasiliev, 2021-04-27:
Pointers only interact with the Unity.UI interface system.
Pointers do not replace the behavior of the mouse pointer at the system level.

If you want pointers to work with objects inside a WebView, you first need to learn how WebView handles incoming mouse events.

eddyaboumansour, 2021-04-27:
@EugeneVasiliev Okay thank you for your help ! I'll check the WebView scripts

andrey-tsb, 2021-07-29:
Closed due to inactivity, please feel free to reopen or create new in case of any issues.

closed 2021-07-29 by andrey-tsb