3DiVi/nuitrack-sdk #20: Segmentation Fault on Ubuntu 20.04

jimruxton, 2021-05-28:
I installed the latest SDK and get a segmentation fault when trying to run the license tool. I am using a D435 sensor which works fine with realsense-viewer. Also the examples in the realsense SDK work. 

When running /usr/bin/QtNuitrackLicense with gdb I get the following:

```
Reading symbols from /usr/bin/QtNuitrackLicense…
(No debugging symbols found in /usr/bin/QtNuitrackLicense)
(gdb) run
Starting program: /usr/bin/QtNuitrackLicense
[Thread debugging using libthread_db enabled]
Using host libthread_db library “/lib/x86_64-linux-gnu/libthread_db.so.1”.
[New Thread 0x7ffff1ca4700 (LWP 57664)]
[New Thread 0x7fffebfff700 (LWP 57665)]
[New Thread 0x7fffeb7fe700 (LWP 57666)]
[New Thread 0x7fffeae7b700 (LWP 57667)]
[New Thread 0x7fffd5f61700 (LWP 57669)]
[New Thread 0x7fffd4c0c700 (LWP 57670)]
[New Thread 0x7fffcdeed700 (LWP 57671)]
[New Thread 0x7fffcd197700 (LWP 57672)]
[New Thread 0x7fffc7fff700 (LWP 57673)]
[New Thread 0x7fffc77fe700 (LWP 57674)]

Thread 6 “Thread (pooled)” received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffd5f61700 (LWP 57669)]
_IO_fgets (buf=buf@entry=0x7fffd5f5e250 “”, n=n@entry=255, fp=fp@entry=0x0)
at iofgets.c:47
47 iofgets.c: No such file or directory.
(gdb) Quit

```
I can’t determine from this what the problem is. Can anyone offer suggestions what to try. Also if I set QT_DEBUG_PLUGINS=1 and run /usr/bin/nuitrack I get the following ( just posting the last few lines) . At the bottom is the seg fault.
```

Snip----------
QElfParser: ‘/usr/bin/zff2gff3.pl’ is not an ELF object
“’/usr/bin/zff2gff3.pl’ is not an ELF object”
not a plugin
QFactoryLoader::QFactoryLoader() looking at “/usr/bin/zip”
“Failed to extract plugin meta data from ‘/usr/bin/zip’”
not a plugin
QFactoryLoader::QFactoryLoader() looking at “/usr/bin/zipcloak”
“Failed to extract plugin meta data from ‘/usr/bin/zipcloak’”
not a plugin
QFactoryLoader::QFactoryLoader() looking at “/usr/bin/zipdetails”
QElfParser: ‘/usr/bin/zipdetails’ is not an ELF object
“’/usr/bin/zipdetails’ is not an ELF object”
not a plugin
QFactoryLoader::QFactoryLoader() looking at “/usr/bin/zipgrep”
QElfParser: ‘/usr/bin/zipgrep’ is not an ELF object
“’/usr/bin/zipgrep’ is not an ELF object”
not a plugin
QFactoryLoader::QFactoryLoader() looking at “/usr/bin/zipinfo”
“Failed to extract plugin meta data from ‘/usr/bin/zipinfo’”
not a plugin
QFactoryLoader::QFactoryLoader() looking at “/usr/bin/zipnote”
“Failed to extract plugin meta data from ‘/usr/bin/zipnote’”
not a plugin
QFactoryLoader::QFactoryLoader() looking at “/usr/bin/zipsplit”
“Failed to extract plugin meta data from ‘/usr/bin/zipsplit’”
not a plugin
QFactoryLoader::QFactoryLoader() looking at “/usr/bin/zjsdecode”
“Failed to extract plugin meta data from ‘/usr/bin/zjsdecode’”
not a plugin
QFactoryLoader::QFactoryLoader() looking at “/usr/bin/zoe-loop”
“Failed to extract plugin meta data from ‘/usr/bin/zoe-loop’”
not a plugin
QFactoryLoader::QFactoryLoader() looking at “/usr/bin/zoom”
“Failed to extract plugin meta data from ‘/opt/zoom/ZoomLauncher’”
not a plugin
QFactoryLoader::QFactoryLoader() checking directory path “/usr/etc/nuitrack/qt/plugins/platformthemes” …
QFactoryLoader::QFactoryLoader() checking directory path “/usr/bin/platformthemes” …
QFactoryLoader::QFactoryLoader() checking directory path “/usr/etc/nuitrack/qt/plugins/platforminputcontexts” …
QFactoryLoader::QFactoryLoader() checking directory path “/usr/bin/platforminputcontexts” …
QFactoryLoader::QFactoryLoader() checking directory path “/usr/etc/nuitrack/qt/plugins/imageformats” …
QFactoryLoader::QFactoryLoader() checking directory path “/usr/bin/imageformats” …
QFactoryLoader::QFactoryLoader() checking directory path “/usr/etc/nuitrack/qt/plugins/iconengines” …
QFactoryLoader::QFactoryLoader() checking directory path “/usr/bin/iconengines” …
QFactoryLoader::QFactoryLoader() checking directory path “/usr/etc/nuitrack/qt/plugins/accessible” …
QFactoryLoader::QFactoryLoader() checking directory path “/usr/bin/accessible” …
QFactoryLoader::QFactoryLoader() checking directory path “/usr/etc/nuitrack/qt/plugins/accessiblebridge” …
QFactoryLoader::QFactoryLoader() checking directory path “/usr/bin/accessiblebridge” …
Segmentation fault (core dumped)
```
I've asked for help on the forum but no luck. Hoping I can find some support here.

VadimAspirin, 2021-11-26:
Hi @jimruxton, what USB version is displayed in the Realsense Viewer? Does the nuitrack app work correctly without a sensor connected?

VadimAspirin, 2022-03-03:
I'm closing this issue, it has been resolved [here](https://github.com/3DiVi/nuitrack-sdk/issues/24#issuecomment-980028615).

closed 2022-03-03 by VadimAspirin