3DiVi/nuitrack-sdk #21: Jittery Results in Unity SDK

snk123, 2021-06-12:
I have used all tutorials and the results are nowhere near the NUITrack platform executable. The skeleton bones keep on disappearing even when staying still. However, when I run the executable and test from there, the results are smooth. (using RealSense D455)

What can be the reason?

emilberzen, 2021-06-15:
Getting the same result with the RealSense 435i. Haven't found a solution yet.

reuksv, 2021-06-28:
Hello @snk123. Try running the "RGB and skeletons" tutorial scene in Unity and then nuitrack_sample.exe from "Nuitrack\nuitrack\ nuitrack\bin"
Is there a difference in joint jitter? If there is a difference, then record the screen.
And write a version of the unity and nuitrack.

reuksv, 2021-10-14:
Case closed due to no further comments received.

closed 2021-10-14 by reuksv

atyshka, 2022-02-09:
I have similar results in the C++ SDK. Here is a recording with RegressionSkeletonization:

https://user-images.githubusercontent.com/19317207/153277390-dbc9f118-1d64-43ec-b94f-dc39783e50a2.mp4

And here is a recording with CNN_HPE:

https://user-images.githubusercontent.com/19317207/153277450-1190a32a-3f37-4ce3-a4e1-97d51ba4a715.mp4