3DiVi/nuitrack-sdk #22: Uninstalling nuitrack throws following "error"

TandersT, 2021-06-24:
C:\WINDOWS\system32>reg add "HKLM\SYSTEM\CurrentControlSet\Control\Session Manager\Environment" /v Path /t REG_EXPAND_SZ /d "%SystemRoot%\system32;%SystemRoot%;%SystemRoot%\System32\Wbem;%SYSTEMROOT%\System32\WindowsPowerShell\v1.0\;%SYSTEMROOT%\System32\OpenSSH\;C:\Program Files\OpenNI\Bin64;C:\ProgramData\chocolatey\bin;C:\Program Files (x86)\ZeroTier\One\" /f 
Value Path exists, overwrite(Yes/No)?

VadimAspirin, 2021-06-30:
Hi @TandersT,

It looks like an uninstaller bug in the old Nuitrack version. You can uninstall Nuitrack manually as follows:
- remove the path to Nuitrack from the PATH environment variable 
- remove the Nuitrack directory (located in the %NUITRACK_HOME% directory)
- remove the NUITRACK_HOME environment variable

andrey-tsb, 2021-07-19:
Closed due to inactivity.  
Please feel free to create new issue in case of any questions.

closed 2021-07-19 by andrey-tsb

941631877, 2022-06-20:
ERROR: Sample failed. Exit code: 3