3DiVi/nuitrack-sdk #23: WIN10 Video Source Selection Window at C# Example Start

emanuelgollob, 2021-09-10:
Hello there, I am using the Nuitrack SDK C# Example with a Nuitrack AI license on a WIN10 PC with a Intel Realsense D455 and I would like to start it automatically with every reboot. 

It works occasionally but very often a WIN10 video source selection window appears and I manually have to choose between the Intel Realsense D455 depth or RGB camera - which is quite annoying as it ideally should reboot without any manual input.

If there is any way to avoiding this window please let me know. 
Cheers

VadimAspirin, 2021-10-05:
Hi @emanuelgollob,

We fix a similar issue with the pop-up window on windows systems on Nuitrack v0.35.9. What Nuitrack version do you use?

emanuelgollob, 2021-10-05:
Hi @VadimAspirin 
I had v0.35.8 installed. After reading your message I updated to Nuitrack v0.35.9 - now it perfects perfectly!
Thank you so much for pointing to the new version!

closed 2022-03-24 by VadimAspirin