3DiVi/nuitrack-sdk #24: Segmentation Fault

Albert950207, 2021-09-18:
Hi, I am new to the nuitrack-sdk

I followed the installation instructions to install nuitrack sdk on ubuntu 20.04 however after I installed and run nuitrack on terminal I get this error:

[image]

Albert950207, 2021-09-18:
Does anyone know what happened ?

Thanks!

VadimAspirin, 2021-09-20:
Hi @Albert950207,

Please try to execute the nuitrack_sample application and send us the output.

Albert950207, 2021-09-24:
Hi @VadimAspirin  

Thank you for your replying. Where is nuitrack_sample application is ?

VadimAspirin, 2021-09-27:
@Albert950207 , you can run nuitrack_sample from the command line as the nuitrack application (just in case: _/usr/bin/nuitrack_sample_).

Albert950207, 2021-09-28:
@VadimAspirin  I run nuitrack_sample from the command line and I get this error:
nuitrack_sample: error while loading shared libraries: libglut.so.3: cannot open shared object file: No such file or directory

VadimAspirin, 2021-09-29:
@Albert950207 please try to install freeglut3 and libbluetooth3 packages and execute the nuitrack app again: `apt install freeglut3 libbluetooth3`

Albert950207, 2021-09-29:
@VadimAspirin   I have installed freeglut3 and libbluetooth3 but after I run nuitrack_sample, I just get a flash of this application and then still get the error:
[image]

VadimAspirin, 2021-09-30:
@Albert950207, please advise, what Nuitrack version and RGBD-sensor model do you use?

VadimAspirin, 2021-09-30:
If you execute nuitrack_device_api_sample and post the output here it will be very helpful too.

Albert950207, 2021-09-30:
@VadimAspirin My RGBD-sensor is intel D435i and I dont know what my version of Nuitrack is but it could be the latest one because I just download it from repository of installation Nuitrack.  I dont know how to check the version because I cant run nuitrack, nuitrack_sample or even nuitrack_device_api_sample. All of them give me an same error: Segmentation fault.

Btw,  I install and run nuitrack in my Windows operating system with same computer and same RGBD-sensor but when I run some samples in nuitrack I got error code 2. I check it on google and google told me that I need to make my intel sensor connected as usb 3.2(mine is 2.1).  Can nuitrack work when intel sensor connected as usb 2.1?

VadimAspirin, 2021-10-05:
@Albert950207, Nuitrack only supports Intel Realsense sensors when Intel Realsense is connected to USB 3 (as Intel recommends).

VadimAspirin, 2021-11-26:
Hi @jimruxton, if I understood correctly [this](https://community.nuitrack.com/t/segmentation-fault-in-ubuntu-20-04/2726) is your issue on the Nuitrack community forum? According to the information I have, removing `libSenDuck.so` from the `/usr/local/lib/nuitrack/OpenNI2/Drivers/` folder helped solve your segmentation fault.

jimruxton, 2021-11-26:
Yes this is correct. It appears |libSenDuck.so |was causing the 
segmentation fault. Removing this driver fixed it.
|
|

VadimAspirin, 2022-04-05:
Hi @Albert950207,

How are you? Has your issue been solved?
Would be great if you could provide some reply/feedback, we will be ready to help.
Thanks.

closed 2022-04-21 by VadimAspirin