3DiVi/nuitrack-sdk #30: tbb.dll error when starting up the project with Unity ver 2020.3.24f1

myuunosuke, 2021-12-10:
When I try to start unity project with nuitrack package on Unity 2020.3.24f1, 
I get message saying...
"You need to replace the tbb.dll file in editor with nuitrack compatible tbb.dll file..."
I replaced the tbb.dll file with the one in the Nuitrack bin folder, and still getting the same error message.
I also did uninstall and install the nuitrack and unity several time but still the same result.

Then I changed to unity ver 2020.3.0f1.
and followed the same steps as above and worked perfectly...

Which version of Unity is the latest supported version?

reuksv, 2021-12-13:
Hello @myuunosuke 
Must work in all. 2020.3.1 and later. It definitely works for us in 2020.3.1, 2020.3.18, 2021.1+ and 2021.2+

junseobma, 2021-12-14:
I also have the same issue.
I tried with version 2020.3.14f1. and version 2020.3.23f1., but it keeps saying "You need to replace the tbb.dll file...".
Obviously I replaced tbb.dll, but it didn't solve the problem.

Also I tried with the 2020.3.0f1. version and I'm still having the same problem.

reuksv, 2021-12-15:
> I also have the same issue. I tried with version 2020.3.14f1. and version 2020.3.23f1., but it keeps saying "You need to replace the tbb.dll file...". Obviously I replaced tbb.dll, but it didn't solve the problem.
> 
> Also I tried with the 2020.3.0f1. version and I'm still having the same problem.

Answered: https://github.com/3DiVi/nuitrack-sdk/issues/31

closed 2021-12-22 by reuksv