3DiVi/nuitrack-sdk #31: tbb.dll error

junseobma, 2021-12-14:
________________________________________________________________________________________
You need to replace the tbb.dll file in Editor with Nuitrack compatible tbb.dll file.
If you click [Yes] the editor will be restarted and the file will be replaced automatically
(old tbb-file will be renamed to tbb_backup.dll)
________________________________________________________________________________________

When I add the Unity package to the Unity project, it tells me to replace tbbdll. 
If I click Yes, it keeps telling me to replace it, and I press Yes endlessly.

Unity project cycled on and off.
Also endlessly telling me to replace tbbdll.

I followed the installation order of nuitrack-sdk/Unity3D/README.md.
1. Platforms/nuitrack-windows-x64.exe
2. Unity3D/NuitrackSDK.unitypackage

I also copied and replaced tbbdll from nuitrack/bin.

nuitrack\bin\nuitrack_sample.exe. This works.

* windows 10
* unity version 2020.3.14f1, 2020.3.23f1, 2020.3.0f1
* Nuitrack v0.35.12

How can I solve this?

reuksv, 2021-12-15:
Hello @junseobma 
Automatic replacement may not work if the user does not have enough rights to access the folder with the editor (For example, \Unity\2020.3.1f1\Editor).
Can you create text-file in Unity Editor folder? Try allow Full control permissions for Users
More details: winaero.com/how-to-take-ownership-and-get-full-access-to-files-and-folders-in-windows-10/)

junseobma, 2021-12-15:
thank you.
It was solved by manually creating a tbb_backup.dll file and replacing it.
It works fine.

closed 2021-12-15 by reuksv