3DiVi/nuitrack-sdk #36: Administrator rights needed for Astra+ support

wldevries, 2022-02-25:
We tried out the Orbbec Astra+ support of Nuitrack version 0.35.12 today. We tested this on a machine with Windows 8.1, the lower compatibility bar for our products. None of the samples work when run as a normal user. The `nuitrack_sample.exe` program for instance exits with the following error:
```
WARNING: OpenNI2DepthProvider can't create RGB Stream
Create DepthProvider: OpenNI2DepthProvider
Access is denied.
terminate called after throwing an instance of 'boost::interprocess::interprocess_exception'
  what():  Access is denied.

This application has requested the Runtime to terminate it in an unusual way.
Please contact the application's support team for more information.
```
They do work when you start them as administrator. The same is true for our application using the Nuitrack SDK. This is different from the Orbbec Astra and Orbbec Astra Pro which can be used in normal user mode. Having to run our applications as administrator is undesirable. Is there something I might have configured wrongly or is there a way to mitigate the problem.

wldevries, 2022-02-25:
> Hi @wldevries,
> 
> Did I understand correctly, on windows 10 astra+ works without administrator rights?

You understood that correctly.

ir4kli09, 2022-02-28:
Have you installed Nuitrack with administrator rights on Windows 8?

logisticAKB, 2022-03-30:
Hi @wldevries 

Have you had chance to try suggestion above? Does the problem still persist?

closed 2022-04-14 by VadimAspirin