3DiVi/nuitrack-sdk #38: Visual C++ Build FAILED

Tripcatmusic, 2022-03-07:
When I try to build, apperas the next error.

Build started...
1>------ Build started: Project: NuiSample, Configuration: Development_Editor x64 ------
1>Invalidating makefile for NuiSampleEditor (NuiSample.Build.cs modified)
1>UnrealBuildTool : error : Unable to determine UHT module type for C:\Program Files\Epic Games\UE_4.25\Engine\Plugins\NuitrackPlugin\Source\NuitrackModule\NuitrackModule.Build.cs
1>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.MakeFile.Targets(46,5): error MSB3073: The command ""C:\Program Files\Epic Games\UE_4.25\Engine\Build\BatchFiles\Build.bat" NuiSampleEditor Win64 Development -Project="C:\Users\THRIA ESTUDIOS\Documents\Unreal Projects\NuiSample\NuiSample.uproject" -WaitMutex -FromMsBuild" exited with code 6.
1>Done building project "NuiSample.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

ir4kli09, 2022-03-10:
Hi @Tripcatmusic,

- Which version of Nuitrack are you using?
- Try to build the project in UE4.
- Have you completed all the steps from the [tutorial](https://github.com/3DiVi/nuitrack-sdk/blob/master/doc/Unreal_Engine_Windows_Tutorial.md#using-nuitrack-in-unreal-engine-425-windows)?
- At what step did you get this error?

Tripcatmusic, 2022-03-12:
Hi
Thanks for answering

UE 4.25.4
When i try to build the Visual studio project appears that error in the
console

[image: image.png]

Tripcatmusic, 2022-03-17:
> Hi @Tripcatmusic <https://github.com/Tripcatmusic>,
>
> Could you please resend your image.

Sure, here it is.

logisticAKB, 2022-03-23:
Hi @Tripcatmusic,

Try to use most recent version of Nuitrack. Does the problem persist? If it does, could you send us sample of your project, so we could investigate the issue?

logisticAKB, 2022-04-05:
Hi @Tripcatmusic 

Has your issue been solved? Do you have any other questions that I can help you with?

Tripcatmusic, 2022-04-05:
Hi, I couldn't solve the problem, I don't know why is that happening

logisticAKB, 2022-04-14:
Hi @Tripcatmusic 
Could you send us sample of your project, so we could investigate the issue?

closed 2022-05-11 by VadimAspirin

reuksv, 2024-02-06:
For someone, this question is still relevant, they have new information.
We conducted additional research and found out that in some cases this happens because the compiler cannot determine the type of the "NuitrackModule" module. Try adding the module manually to "NuitrackPlugin.plugin"

	

	"Modules": [
		{
			"Name": "NuitrackPlugin",
			"Type": "Runtime",
			"LoadingPhase": "Default"
		},
		{
			"Name": "NuitrackModule",
			"Type": "Runtime"
		}
	]