3DiVi/nuitrack-sdk #39: Nuitrack Linux error - Please run as current user

tomassilny, 2022-05-08:
Hi,
I am getting this error message when I type `nuitrack` after installing the latest Nuitrack SDK on Orange Pi:

[image]

I tried also `sudo nuitrack` with the same result.

My specifications:
HW: OrangePi 3 LTS
OS: Debian buster 2.1.6
ARCH: arm64
NUITRACK SDK VERSION: [v0.35.14](https://github.com/3DiVi/nuitrack-sdk/commit/e0a0005d83827c12215f505eb764249ac4173a15)

Thank you for any tips!

logisticAKB, 2022-05-16:
Hi @tomassilny 

You should run Nuitrack as a non-root user

logisticAKB, 2022-05-23:
Hi @tomassilny 

Has your issue been solved? Do you have any other questions I can help you with?

logisticAKB, 2022-05-25:
Hi @tomassilny 

How do you use your OrangePi? Are you connected via SSH or directly with monitor connected to OrangePi?

tomassilny, 2022-05-25:
@logisticAKB I am connected via SSH

tomassilny, 2022-05-30:
Okay, that was a problem, I installed VNC viewer and now it works! Thank you

closed 2022-05-30 by tomassilny