3DiVi/nuitrack-sdk #41: Qt_5.11 still not found during activation

crystal-butler, 2022-05-11:
I have the same issue under Ubuntu 18.04 as @bearpaw in [issue #6](https://github.com/3DiVi/nuitrack-sdk/issues/6#). I downloaded the nuitrack installer just now, so presumably have the latest version of the application.

_Originally posted by @crystal-butler in https://github.com/3DiVi/nuitrack-sdk/issues/6#issuecomment-1123763188_

VadimAspirin, 2022-05-25:
Hi @crystal-butler,

I can recommend this command as a workaround to run the nuitrack activation tool:
```
LD_LIBRARY_PATH=$NUITRACK_HOME/qt/lib:$LD_LIBRARY_PATH nuitrack
```

VadimAspirin, 2022-06-01:
Hi @crystal-butler,

How are you? Has your issue been solved? 
Would be great if you could provide some reply/feedback, we will be ready to help.
Thanks.

closed 2022-06-10 by VadimAspirin