3DiVi/nuitrack-sdk #42: Is it possible to pass video & depth data directly to the SDK, i.e. from a custom device?

dbruning, 2022-06-17:
I understand that the SDK has support for multiple devices. If I have an unsupported device, can I pass depth & color data frame-by-frame to the SDK manually? i.e. can I implement some device interface myself?

andrey-tsb, 2022-06-17:
Hi @dbruning, it is not supported in the public revision of Nuitrack due to the following reasons:
1. Nuitrack relies not only on the image/depth data, but additional parameters of specific depth sensor model (which are requested directly from the driver of depth sensor).
2. Nuitrack mostly licensed per sensor, so the sensor in this case performs as a kind activation key/token (like a USB protection token). 

But we have been providing different custom revisions of Nuitrack (with different licensing models) to our major customers, so if you have any specific volume demand - please feel free to contact us (through feedback form on nuitrack.com or directly at nuitrack@3divi.com).

andrey-tsb, 2022-06-17:
Also if you have any unsupported sensor on hands, please let us know the model of this sensor and we include it into our roadmap. (in principle, any feature in the roadmap could be prioritized through a custom development process)