3DiVi/nuitrack-sdk #46: Object detection not working

esbennn, 2022-08-15:
Hi,
I'm trying to get Object detection working for a project.
The camera is a Realsens D435 and the platform is windows 10. 
With the nuitrack activation tool, I have activated an AI license for the camera, and the test samples work - unless I check the "AI: Object Detecion", then it opens a window and immediately closes it again, with the following message in the box:
```
Starting test...
    3D sensor serial number: 215322076948
    3D sensor provider name: Realsense2DepthProvider
ERROR: Sample failed. Exit code: 3
```
Unckecking the AI: Object Detection box, it works as expected

As detailed in [the online docs](https://community.nuitrack.com/t/nuitrack-video-tutorial-2-object-detection-with-nuitrack-ai/2501), I have installed Cuda 9

```
C:\Users\Playground>nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2017 NVIDIA Corporation
Built on Fri_Sep__1_21:08:32_Central_Daylight_Time_2017
Cuda compilation tools, release 9.0, V9.0.176
```
The PC has a NVIDIA GeForce RTX 2060 with latest drivers installed.

I have tried installing an older version of Nuitrack (v0.35), with the same results. I intended to install v0.34, as that is th version used in the video above, but executables for older versions are not available.

Any ideas?

logisticAKB, 2022-08-18:
Hi @esbennn,

Please try to execute the application by path: %NUITRACK_HOME%\bin\nuitrack_sample.exe and send us the output.

esbennn, 2022-08-18:
```
C:\Users\Playground>"%NUITRACK_HOME%\bin\nuitrack_sample.exe"
WARNING: Cannot load library module: C:\Program Files\Nuitrack\nuitrack\nuitrack/middleware/NuitrackModuleGPU.dll
WARNING: Cannot load library module: C:\Program Files\Nuitrack\nuitrack\nuitrack/middleware/NuitrackModuleGPU.dll
terminate called after throwing an instance of 'tdv::nuitrack::TerminateException'
  what():  WARNING: Cannot load library module: C:\Program Files\Nuitrack\nuitrack\nuitrack/middleware/NuitrackModuleGPU.dll

C:\Users\Playground>
```
The `NuitrackModuleGPU.dll` seems to exist at the specified location:
```
C:\Program Files\Nuitrack\nuitrack\nuitrack\middleware>cd "%NUITRACK_HOME%\middleware\"

C:\Program Files\Nuitrack\nuitrack\nuitrack\middleware>dir
 Volume in drive C is OS
 Volume Serial Number is DA13-B5EA

 Directory of C:\Program Files\Nuitrack\nuitrack\nuitrack\middleware

08/15/2022  08:16 AM    <DIR>          .
08/15/2022  08:16 AM    <DIR>          ..
11/25/2020  01:37 PM         8,148,992 NuitrackModule.dll
11/25/2020  01:37 PM         8,178,688 NuitrackModuleGPU.dll
11/25/2020  01:37 PM           909,824 OpenNIModule.dll
               3 File(s)     17,237,504 bytes
               2 Dir(s)  209,679,962,112 bytes free

C:\Program Files\Nuitrack\nuitrack\nuitrack\middleware>
```

esbennn, 2022-08-31:
Okay, after trying out on a different machine (linux), I managed to install the (ancient versions of ) cuda and associated cdnn files. This worked.
I then tried reinstalling everything on the windows machine, and it worked. Maybe the cdnn files were somehow corrupted on download the first time.

closed 2022-08-31 by esbennn