3DiVi/nuitrack-sdk #47: 使用unity开发，在处理器为i3 Windows平台运行发生错误

lichenyigit, 2022-08-30:
使用unity开发，在处理器为i3 Windows平台运行发生错误，具体报错如下：
```
A crash has been intercepted by the crash handler. For call stack and other details, see the latest crash report generated in:
 * C:/Users/windows/AppData/Local/Temp/Unity/Editor/Crashes

=================================================================
	Native Crash Reporting
=================================================================
Got a UNKNOWN while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
=================================================================

=================================================================
	Managed Stacktrace:
=================================================================
	  at <unknown> <0xffffffff>
	  at nuitrack.NativeImporter:nuitrack_CreateUserTracker <0x0008a>
	  at nuitrack.NativeUserTracker:.ctor <0x00042>
	  at nuitrack.UserTracker:Create <0x00052>
	  at NuitrackManager:NuitrackInit <0x00d2a>
	  at NuitrackManager:FirstStart <0x00222>
	  at NuitrackManager:Awake <0x000ea>
	  at System.Object:runtime_invoke_void__this__ <0x00087>
=================================================================
```

lichenyigit, 2022-08-30:
一下是unity 日志内容:
```
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                          [... cut ...]
0x00007ff781e74e7f (Unity) PreloadManager::WaitForAllAsyncOperationsToComplete
0x00007ff78306b417 (Unity) EditorSceneManager::RestoreSceneBackups
0x00007ff782d79905 (Unity) PlayerLoopController::EnterPlayMode
0x00007ff782d88ea7 (Unity) PlayerLoopController::SetIsPlaying
0x00007ff782d8c2ad (Unity) Application::TickTimer
0x00007ff7831dadba (Unity) MainMessageLoop
0x00007ff7831df66b (Unity) WinMain
0x00007ff78458475e (Unity) __scrt_common_main_seh
0x00007ffa0feb7034 (KERNEL32) BaseThreadInitThunk
0x00007ffa10d22651 (ntdll) RtlUserThreadStart

```

lichenyigit, 2022-08-30:
而且发现，笔记本使用nuitrack 也是直接闪退。
笔记本配置：
```
win11专业版 ，设备名称	thinkpad-Binttech
处理器	Intel(R) Core(TM) i5-10210U CPU @ 1.60GHz   2.11 GHz
机带 RAM	16.0 GB (15.7 GB 可用)
设备 ID	6386EC98-DDBF-40B1-9448-F36EAA10BFEA
产品 ID	00331-20020-00000-AA549
系统类型	64 位操作系统, 基于 x64 的处理器
笔和触控	没有可用于此显示器的笔或触控输入
显卡：NVIDIA GeForce MX250
```

reuksv, 2022-09-01:
Hello @lichenyigit . Does it work Nuitrack\nuitrack\nuitrack\bin\nuitrack_sample.exe? Have other skeleton trackers been installed before? (for example cubemos)

reuksv, 2022-09-08:
> Hello @lichenyigit . Does it work Nuitrack\nuitrack\nuitrack\bin\nuitrack_sample.exe? Have other skeleton trackers been installed before? (for example cubemos)

@lichenyigit ?

reuksv, 2022-09-20:
Well, I hope the problem has disappeared. If not, then feel free to reopen the ticket

closed 2022-09-20 by reuksv

lichenyigit, 2022-10-02:
问题已经解决了。解决方案：将unity打包好的桌面程序，在运行的时候一定要是管理员角色。否则就会报这个问题。我尝试了，只有在使用nuitrack插件的情况下才需要这样操作。至今没有搞明白为什么需要这么操作。。