3DiVi/nuitrack-sdk #52: unity平台，nuitrack启动 CPU跑到了100%

lichenyigit, 2022-10-02:


lichenyigit, 2022-10-02:
对了，还有一个问题。unity平台打包出Windows应用程序后，在启动时nuitrack从初始化到完全运行的过程特别长，大概40秒左右。请问这个问题怎么解决

reuksv, 2022-10-05:
1. What type of tracking do you use (standard or AI)?
2. Does this problem persist in the sample Nuitrack\nuitrack\nuitrack\bin\nuitrack_sample.exe?

lichenyigit, 2022-10-05:
感谢你的回复。
我的程序是在Windows平台上运行的；
第二个问题，没有明白，是问我用的哪种类型的lisence吗？lisence类型是Nuitrack AI Perpetual
运行Nuitrack\nuitrack\nuitrack\bin\nuitrack_sample.exe CPU也是跑满
[image]

logisticAKB, 2022-10-10:
Hi @lichenyigit 

Regarding second question: we need to know what skeletonization type you use. You can check it in nuitrack.config: Skeletonization.Type field or you can tell us output of Debug.Log from your first screen

lichenyigit, 2022-10-11:
[image]
你好，这个是截图。

lichenyigit, 2022-10-22:
你好，CPU运行100%的情况通过修改配置文件解决了，表示感谢。还有一个问题：unity打包为Windows运行程序启动时特别慢 怎么解决？

logisticAKB, 2022-11-15:
Hi @lichenyigit 

Startup time of the Nuitrack mostly depend on number of sensors connected to the PC and their vendor. 
If you have multiple sensors connected you can leave one connected if possible