3DiVi/nuitrack-sdk #53: Could Nuitrack provide 3D hand data?

StevenZzz07, 2022-10-11:
Hi, I am looking for commercial software that can provide 3D hand data. Developer from IntelRealsense said that Nuitrack can provide such data using Realsense device? I want to collect 3D hand info (joints position, 3D hand model). So could Nuitrack meet my needs? If so, we mayt consider to purcase the softwar3. Please reply to me soon, thanks!

Vizonary, 2022-10-14:
We asked the same from NuiTrack and they don't exactly have something like that in the pipeline but only for personal or enterprise use. How we solved this, was by using a different Computer Vision program and running it simultaneously.

andrey-tsb, 2022-11-07:
Hi, to provide a kind of official response I should refer to [an answer](https://community.nuitrack.com/t/feature-request-can-this-track-hand-movements-and-finger-joints/3023/2?u=tag) to the similar question asked in [community forum](community.nuitrack.com).

Specifically for Realsense - it is hardly feasible, unfortunately, due to specific noise distortions of stereo sensors.
The only exception is tracking with a very small distance from the sensor to the hand.
Could you tell please what working distance do you suppose for your use-case ?

Vizonary, 2022-11-08:
Hey,
Thanks for replying. The distance would be 20cm to 60cm for the 3D hand
tracking.
Thanks!

closed 2023-01-26 by andrey-tsb