3DiVi/nuitrack-sdk #58: cannot select my kinect v1

jingerbreadBiliBili, 2023-05-07:
I got a kinect v1 and i try to use a another program to do 3D dynamic capture to a model ,it use nuitrack but it cannot discover my kinect

jingerbreadBiliBili, 2023-05-07:
there is a normal sdk (v1.6) on my computer
i don't know will is that reason

jingerbreadBiliBili, 2023-05-11:
whatever

logisticAKB, 2023-05-17:
Hi @jingerbreadBiliBili 

Could you describe the problem in details?

- Is your kinect v1 detected by latest Nuitrack.exe?
- What software do you use that cannot detect your sensor? How is it related to Nuitrack?
- What do you mean by  "normal sdk (v1.6)"?

logisticAKB, 2023-05-26:
Hi @jingerbreadBiliBili 

We are interested in understanding the problem and helping you solve it. 
Could you clarify to us the details requested above?

closed 2023-06-08 by VadimAspirin