3DiVi/nuitrack-sdk #59: Using nuitrack on pre-recorded videos/point clouds

benpocklingtonhes, 2023-09-28:
Hi, I have a large number of recordings from the Azure Kinect DK from a project (In the .mkv format that Microsoft use for the Kinects), and I was wondering if nuitrack supports skeleton generation using pre-recorded data?

Our problem was that we had issues with multiple cameras and generating skeletons for them at the same time (with the native Microsoft skeleton generation). The recordings contain all the relevant settings and parameters for the Kinect used in the metadata. Please let me know whether it is currently possible, even if I have to code a workaround or two.

andrey-tsb, 2023-10-02:
Hi, thank you for your questions.  
Azure mkv recordings aren't readily supported by Nuitrack. We have only limited support for *.oni/*.bag formats in Trial version. Now we don't have any particular plans for extended support of pre-recorded sequences due to (1) low demand (2) uncertain business/licensing model of such tracking mode.

benpocklingtonhes, 2023-10-09:
Ok thank you for the information!

closed 2023-10-09 by benpocklingtonhes