3DiVi/nuitrack-sdk #61: PythonNuitrack example on Readme crashes when face tracking enabled

eleKit, 2023-11-09:
When I run the example in the readme of your python wrapper everything works well until I enable the two lines of code for face detection:
nuitrack.set_config_value("Faces.ToUse", "true")
nuitrack.set_config_value("DepthProvider.Depth2ColorRegistration", "true")

When I run the code forface detection, as soon as a face is detected, it crashes and returns this error:

Traceback (most recent call last):
  File "C:\Users\me\my\Pah\main.py", line 67, in <module>
    data_instance=nuitrack.get_instance()
RuntimeError: No such node (quality)

I'm using windows 10, the py_nuitrack_windows_python3.10-0.0.1-py3-none-any.whl and the Intel RealSense D435

logisticAKB, 2023-11-20:
Hi @eleKit 

We've confirmed a bug on our side. We'll fix this in upcoming release. Sorry for the inconvenience

logisticAKB, 2024-01-15:
Hi @eleKit 

Issue was fixed in Nuitrack v0.37.7 - https://github.com/3DiVi/nuitrack-sdk/releases/tag/v0.37.7

eleKit, 2024-01-15:
Thank you very much!

closed 2024-01-15 by eleKit