3DiVi/nuitrack-sdk #62: [Feature Request] Ability to control and force detection in a specific area, in depth or in space

berkut0, 2023-11-12:
I love your tool, it saves me weeks of work. But I have the same problem every time - if the stage is not meant to be used in a specially adapted room, and the device itself is facing the audience, there is a problem of false positives and I have to get clever with the filtering.

Is it possible to expect "smart filtering"? For example, I would be happy to cut off users "no further than a certain line". But it would be ideal to be able to set priority areas in the room to block some zones.

At the moment, this feature is the most problematic for almost any usage scenario.

VadimAspirin, 2023-11-16:
Hi @berkut0,

Currently, you can limit the maximum tracking distance using the `Skeletonization.MaxDistance` parameter, which you can change in the `%NUITRACK_HOME%/data/nuitrack.config` file.
A more detailed configuration of the workspace is not available at the moment and is on our roadmap, but so far without an exact implementation date, mainly due to low demand.

andrey-tsb, 2023-11-16:
Hi @berkut0, nice to meet you, actually "smart filtering" (annotated smart 3d space) is the important subject of the development for our **Nuitrack Holistic**. 

Could you describe a little bit some typical definitions ("block some zones") which are useful for your application/project ?  
What programming language/environment do you use currently - Unity, Unreal Engine, Python, something else ?

If it's more convenient - please feel free to reach us directly at nuitrack@3divi.com.

berkut0, 2023-11-17:
Primarily, I am working with the implementation of your library for [VVVV Gamma](https://visualprogramming.net/).
Less often I recommend using it to colleagues on Unity.

Probably the easiest way to trim zones is with primitives. But I work a lot with Signed Distance Fields (SDF) and I must say that this approach gives much more flexibility in "designating" the space.

I also find it important to designate zones with increasing priority (e.g. by distance), because in scenarios where you want to identify users by proximity to a point or zone this is crucial. Especially in the context of multi-user scenarios. In such a case, SDF can be a very important tool.

@VadimAspirin What you're talking about is amazing! I will definitely try it!

I'll get back to you via email, thanks.

berkut0, 2023-11-17:
> Currently, you can limit the maximum tracking distance using the `Skeletonization.MaxDistance` parameter, which you can change in the `%NUITRACK_HOME%/data/nuitrack.config` file.

Does it work with body images?

VadimAspirin, 2023-11-20:
> > Currently, you can limit the maximum tracking distance using the `Skeletonization.MaxDistance` parameter, which you can change in the `%NUITRACK_HOME%/data/nuitrack.config` file.
> 
> Does it work with body images?

This only affects the skeletons in the output.

If you are using Intel Realsense, you can affect all output with this similar parameter:  
  
[image]  
  
Is it useful for you?

berkut0, 2023-11-20:
@VadimAspirin Thanks!

In my setups I tend to use Body Contour images rather than skeletons. Skeletons came into my development pipeline when I was trying to implement intelligent filtering. It would be very cool to have the cutting contour with spatial control right out of the box.