3DiVi/nuitrack-sdk #65: Is there necessary to work GPU for nuitrack AI?

kookoo9999, 2024-01-11:
I wanna run nuitrack AI's multiple sensor on mini PC like intel NUC with 3 cameras.

so I tried multiple sensor with 3 cameras on NUC(without GPU) but it doesn't work and I tried that on desktop(cpu: i5-11600KF, GPU :3060, ram 32GB) it works well.

And I tested at my labtop(CPU : 5600U , non extern GPU, ram :16GB) but it doesn't work.

Can I know what is minimum spec for working multiple sensor(3 cameras)??

Thank you for reading . I hope you have a nice day!

andrey-tsb, 2024-01-12:
Hi there, GPU isn't required for AI skeletal tracking, but required for object detection, so the answer depends on the feature-set you are using.

General recommendation in terms of hardware resources for multi-sensor AI skeletal tracking is having 2-4 desktop cores / per input stream (depending on CPU frequency). 

Intel NUC isn't guaranteed to have a 30fps for 3 cameras, but this model is quite capable to launch at least, so if it doesn't work at all - it's not expected behavior.

The same is relevant for your laptop and single input stream. Nuitrack AI was successfully tested on ultrabooks with an Intel CPU (U series) and w/o dedicated GPU. 

Do you have some error messages when testing with NUC and your laptop ?

kookoo9999, 2024-01-23:
@andrey-tsb 
Thank you for your answer.
I've been trying at other desktop and activate the license.
but the following error on each sensors(3 sensors, L515) message appears:Activation error: Server error response: License reactivation with the different sensor does not allowed.
However, I confirmed that the license for each sensor was the same. So I don't know what the cause is.
Can you tell me how to solve this problem?

abiliaorg, 2024-02-03:
How did you activate AI skeleton recognition through SDK? Thanks!

kookoo9999, 2024-02-06:
@abiliaorg 
I tried to activate license with Nuitrack(in activation_tool).
but it doesn't work

andrey-tsb, 2024-02-06:
Sorry for late reply (actually the most straightforward to get help in such cases is to create Help Desk ticket at https://3divi.freshdesk.com/)

>  However, I confirmed that the license for each sensor was the same. So I don't know what the cause is.

We believe there is some confusion here, Nuitrack Perpetual is licensed per depth sensor, so it's bound to a specific sensor during activation. 
To try Nuitrack with 3 sensors you could use a single Nuitrack Trial license or each sensor should be licensed with a Nuitrack Perpetual license.

andrey-tsb, 2024-02-06:
Hi @abiliaorg , license could be activated manually through a Nuitrack.exe or programmatically through an API (please take a look at [Device API](https://github.com/3DiVi/nuitrack-sdk/blob/master/doc/Device_API.md)).

kookoo9999, 2024-02-07:
@andrey-tsb 
Thank you for your answer.
I wonder if I can activate the license for limited time(like a 3 times,, etc)??
I don't know why activating a license is possible on some computers and not on others.

andrey-tsb, 2024-02-16:
Hi @kookoo9999 

> I wonder if I can activate the license for limited time(like a 3 times,, etc)??
> I don't know why activating a license is possible on some computers and not on others.

Which Nuitrack license do you use currently (in the described case) ? 

- Nuitrack Perpetual could be activated with the **same depth sensor** on **any number of computers**. Please let us know through a Help Desk in case of any issues. 
- Nuitrack Trial (license with a limited 3-minutes runtime) could be activated on a **single computer only** and could work with **any support depth sensor**. Also you could issue any number of Nuitrack Trial licenses as you wish.

We hope this helped, but please let us know if uncertainty remained.

andrey-tsb, 2024-02-16:
Also - answering your original question in terms of licenses : 

> Can I know what is minimum spec for working multiple sensor(3 cameras)??

In this case you need 
- 1x Nuitrack Trial license (it will work with multiple sensors, but have a limited 3-minutes runtime)
- **OR** 3x Nuitrack Perpetual license (for each depth sensor) to use multi-sensor tracking without any runtime limitations

andrey-tsb, 2024-02-27:
Hello @kookoo9999 , how are you ? 
We really anticipate any feedback on how this worked for you (or not).

kookoo9999, 2024-02-28:
@andrey-tsb 
Thank you for your answer.
I checked it didn't work. but it works well on previously licensed PCs.

andrey-tsb, 2024-02-28:
We're ready to directly support this case (if its relevant for you), feel free to [create a Help Desk ticket](https://3divi.freshdesk.com/).

closed 2024-03-10 by andrey-tsb