3DiVi/nuitrack-sdk #66: Maximum detection range

machongrun, 2024-01-26:
Your tools are amazing! I want to know what the maximum detection range of the Skeleton tracking SDK is ？
Many thanks！

andrey-tsb, 2024-02-08:
Hi @machongrun, sorry for late reply.
Actually it depends on a specific sensor resolution and field of view (primarily horizontal).
What should be expected on average - 3-4 meters. For some sensors - up to 5-6 meters. 
For proper sensors it could be additionally fine-tuned (with increased CPU/GPU usage) - possibly up to 7-8 meters.
In general, higher working distances aren't feasible on depth sensors.

andrey-tsb, 2024-02-16:
Hi @machongrun , how are you? 
What depth sensor do you currently?
With this information we'll be able to provide a more accurate answer and could assist with Nuitrack fine-tuning.

closed 2024-02-27 by andrey-tsb