3DiVi/nuitrack-sdk #69: Nuitrack App crashing with Astra 2 camera

jp887, 2024-03-26:
This issue relates to my post on the community forum linked below, however the community forum seems to have been down for several days so I have been unable to respond/progress discussion there:
https://community.nuitrack.com/t/nuitrack-app-crashing-with-astra-2/3593

Upon launching the Nuitrack App with an Astra 2 on Windows 11 the app crashes when "Try Nuitrack!" is clicked.
As per @VadimAspirin's suggestion I have tried the OrbbecViewer application. Initially, OrbbecViewer was listing the Astra 2 as a USB 3.0 device and subsequently I was able to use the Nuitrack App in Classical mode sucessfully (AI mode would still crash).

However, since then the OrbbecViewer has started listing the Astra 2 as a USB 2.0 device, and the Nuitrack App is again crashing in both Classical and AI modes. I cannot explain why OrbbecViewer has intermittently listed my camera as USB 3.0 vs USB 2.0, or why Nuitrack App worked sometimes in Classic mode.

I have tried running the OrbbecViewer and Nuitrack App as Administrator, re-installing both Nuitrack and the Orbbec Astra Win32 Driver, however this has not made any difference.

andrey-tsb, 2024-03-26:
Hi @jp887,
We have major availability issues on the hosting side which impacted some of our resource (incl. community forum), sorry for that.

In the meantime, going to the subject - could you please specify your CPU model ? 
We had compatibility issues with some recent CPUs, which are now "almost resolved" (we have a closed beta version, which could be tried out on your side and planned to be released soon).

jp887, 2024-03-26:
Hi @andrey-tsb, I'm running an AMD Ryzen 7 7700 CPU. Thanks.

jp887, 2024-04-11:
I can confirm that this issue was related to Zen 4 processor support, and is fixed for me in the latest release of Nuitrack (0.37.14).

closed 2024-04-11 by jp887