3DiVi/nuitrack-sdk #71: UE5 plugin error

ancillarymagnet, 2024-04-01:
Trying to run the marketplace version of the plugin on UE 5.2 results in an error ...

[image]

This happens when trying to play in editor on any of the levels in the plugin Samples directory, using an Orbbec Femto Bolt on Nuitrack SDK 0.37.13.

reuksv, 2024-04-02:
Hello @ancillarymagnet @pc2555 
We are already working on a solution to this problem. Try using the previous version of Nuitrack (0.37.12)
https://github.com/3DiVi/nuitrack-sdk/blob/v0.37.12/Platforms/nuitrack-windows-x64.exe

ancillarymagnet, 2024-04-02:
I was able to get these examples to work with 0.37.12 thanks

reuksv, 2024-04-09:
@ancillarymagnet @pc2555 
Plugin Updated :rocket: (fixed compatibility with the latest version of Nuitrack)
https://www.unrealengine.com/marketplace/en-US/product/nuitrack-real-time-body-tracking

closed 2024-04-10 by reuksv