3DiVi/nuitrack-sdk #72: TouchDesigner integration not finding device

ancillarymagnet, 2024-04-01:
Using Nuitrack 0.37.13 and TouchDesigner 2023.11510 or 11290, whether:

* Creating a new .toe file and dropping a new Nuitrack CHOP 
or
* opening the provided Nuitrack-Sample.toe

and whether running the Nuitrack demo (Try Nuitrack! button in the Nuitrack app) or not,

... the CHOP shows an error in TouchDesigner:
[image]

ir4kli09, 2024-04-02:
Hello @ancillarymagnet 
Thank you for bringing this issue to our attention. We will test the plugin with version TD 2023 and get back to you with an update as soon as possible.

ir4kli09, 2024-04-03:
Hello @ancillarymagnet 
I tested the plugin on version TD 2023.11220, it works as expected. But I was able to reproduce your problem. Could you tell me if you had another app running in the background using your sensor? If yes, please close it and run the TD sample again.

ancillarymagnet, 2024-04-03:
Hi @ir4kli09 just to confirm I did a test from a reboot, straight into 2023.11290, no other applications running, and still have the issue.

ir4kli09, 2024-04-12:
Hello @ancillarymagnet 
We have solved the problem, we will release an update soon

andrey-tsb, 2024-04-14:
Hi @ancillarymagnet, you are welcome to try the latest release ([0.37.15](https://github.com/3DiVi/nuitrack-sdk/releases/tag/v0.37.15)) for the relevant TouchDesigner compatibility fix.

ancillarymagnet, 2024-04-16:
Hi @andrey-tsb - I installed 0.37.15 and re-downloaded the Nuitrack-sample.toe (in case it was updated? hard to tell at the bit.ly link) and tried TouchDesigner 2023.11290 and 11510 ... still have the same problem as before: "Error: Check if the device is connected and reload the plugin."

closed 2024-05-14 by ir4kli09