3DiVi/nuitrack-sdk #73: Unreal 5.3 w Blueprints?

ancillarymagnet, 2024-04-18:
Is there a way to use the plugin with UE 5.3 Blueprints? Currently the automatic plugin fails on 5.3.2. 

As far as I can tell, there is no zip archive in ({Your project}\Plugins\NuitrackPlugin\Binaries\Win64) as suggested by the 'Use pre-builded libraries' step in the github.

Step 3, which is Manual Rebuilding, only works on a C++ project, if I understand correctly.

reuksv, 2024-04-19:
Hello @ancillarymagnet 
Did you download this plugin from the marketplace?
https://www.unrealengine.com/marketplace/en-US/product/nuitrack-real-time-body-tracking

ancillarymagnet, 2024-04-19:
Yeah the plugin I'm using is downloaded through the marketplace and installed to 5.3.2 editor via the Epic Launcher.

reuksv, 2024-04-23:
@ancillarymagnet 
Try downloading the latest version of Nuitrack 
Create a new project  
Enable the plugin in this project  (in "Plugins")
And run any sample from the Sample folder (see screenshot)
[image]

ancillarymagnet, 2024-04-26:
Hey @reuksv ... not sure what got this straightened out but I no longer have the issue and can run the plugin with 5.3.2. Thanks for your help.

closed 2024-05-02 by reuksv