3DiVi/nuitrack-sdk #77: Hand gestures for Android phones

MDVGames, 2024-07-03:
Hi.

Any plans to support hand gestures for Android phones (like Google MediaPipe)?

andrey-tsb, 2024-07-03:
Hi @MDVGames, 
Aren't you satisfied with the MediaPipe ? Why not just use it in this case ? 

(Nuitrack roadmap isn't set in stone, that's why we always try to work out additional use-cases for Nuitrack through similar questions). 

Thanks.

MDVGames, 2024-07-03:
Currently, the MediaPipe Plugin for Unreal Engine 5 only works on PC + webcam. It does not work on Android phones. 

Would be great if the Nuitrack Plugin for Unreal Engine 5 had the capability to export to Android phones with hand tracking support.

andrey-tsb, 2024-07-04:
Thank you for getting back.
- Nuitrack fully supports Android devices with depth sensors attached (like Orbbec Persee / Persee+ or Android phone with an Orbbec/Intel depth sensor), but there is no support for conventional phone camera.
- Hand could be tracked as a single joint inside full body skeleton and also simple hand gestures are supported, but there is no support for detailed hand pose ("finger tracking") 

We're considering Nuitrack extension in this direction for the long-term, but really don't recommend to rely on this for the nearest projects / plans.

MDVGames, 2024-07-04:
Ok. Thank you for the information.

closed 2024-07-05 by MDVGames