3DiVi/nuitrack-sdk #79: [TouchDesigner Plugin] Tracking multi-person

nariakiiwatani, 2024-12-09:
Hi, I run the TD plugin and I found there is no way to track multiple person.
I see only one person in NuiTrackCHOP when having multiple person in front of RealSense D435.

Is there any way or plan to track multiple person?

ir4kli09, 2024-12-13:
Hello @nariakiiwatani 

At the moment, the plugin only works with one user
> Is there any way or plan to track multiple person?

Yes, but unfortunately I can't say when we will release the update

andrey-tsb, 2024-12-16:
Hi @nariakiiwatani, if your commercial project depends on this feature - please contact us directly through a feedback form (as there are usually options to prioritize it based on the importance of project).

nariakiiwatani, 2024-12-16:
@andrey-tsb Thank you!
But we have already managed this via python script that sends OSC to TD.
Though it will be still nice for us to have new version of TD plugin, it's not necessary for now.

andrey-tsb, 2024-12-19:
@nariakiiwatani Great news! 
Is there any kind of reference implementation for OSC-related part ? 
We're here at Nuitrack are still kind of newbies in TD-stuff - heard multiple times of OSC, but didn't have time yet to get acquainted with it.
If your implementation could be generally applied to Nuirack + TD and shared with other users (kind of semi-official integration package)  - we could provide you with Nuitrack licenses in exchange for this contribution. 
Please let us know if this could be relevant for you.

nariakiiwatani, 2024-12-19:
Great suggestion!
We can provide both parts; python and TD.
Let me have some time to get it ready for sharing :)

nariakiiwatani, 2024-12-20:
@andrey-tsb Please check it out!
https://github.com/nariakiiwatani/Nuitrack-Python-OSC-TouchDesigner

Currently it contains only the minimum range of functions used in our project.
Feel free to open an Issue if there are modifications or additional features needed to make it suitable as a semi-official one.

ir4kli09, 2025-01-10:
Hi @nariakiiwatani 
Sorry for the delay in responding. 
We are looking into your project and similar interactions with TD, we will get back with a response soon.

nariakiiwatani, 2025-02-18:
Hello @ir4kli09
I just wanted to follow up. I’m not trying to rush you—just checking in for confirmation.

If possible, I’d like to have two additional licenses ready within the next two days.
If I can receive them as part of the contribution arrangement, that would be great.
Otherwise, I’m prepared to purchase them.
Either way works for me.

Please let me know at your convenience. Thanks in advance for your time!

andrey-tsb, 2025-02-18:
Hello @nariakiiwatani,
We're sorry for such a long delay 🤦  Thank you for your patience 🙏
We've just added a pack of Nuitrack AI Perpetual licenses to your account - 10 as our gratitude for your contribution and 5 to make amends for our sluggish responses  :-)
We hope this would help your current and future projects. 

P.S. I'll duplicate this message directly in email just to confirm everything arranged correctly and licenses are available on your side. We could also work additional options directly in email in case Perpetual licenses aren't suitable for you.

nariakiiwatani, 2025-02-18:
Thank you so much for your quick and kindest response.
I'm going to reply in email!